greatscottgadgets/greatfet #392: Logic driver issue Windows

the-lifting-engineer, 2021-08-25:
Running into a driver issue when I try to run the logic command on Windows 10. 

gf logic -p /tmp/spi-capture12.sr
Sampling 8 channels at 17 MHz.
Press Ctrl+C to stop reading data from device.

ERROR: Communications failure -- check the connection to -- and state of  -- the GreatFET.
(More debug information may be available if you run 'gf dmesg').
[Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'

Capture terminated -- waiting for data processing to complete.
Sigrok/PulseView compatible session file created: '/tmp/spi-capture12.sr'.
Sampled for 3.0159 seconds.

When I run dmesg it doesn't pull any errors and when I run the gf info command it has no problem detecting the board.

I've reinstalled the drivers through Zadig but it appears to have made no difference.

pickeditmate, 2022-04-29:
I'm getting a similar error with this too. Windows 10

C:\Users\User>gf logic -p testSigrokCapture.sr
Sampling 8 channels at 17 MHz.
Press Ctrl+C to stop reading data from device.
Traceback (most recent call last):
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 236, in get_interface_and_endpoint
    return self._ep_info[endpoint_address]
KeyError: 129

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\Python310\Scripts\greatfet_logic-script.py", line 33, in <module>
    sys.exit(load_entry_point('GreatFET==2021.2.1', 'console_scripts', 'greatfet_logic')())
  File "C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet\commands\greatfet_logic.py", line 303, in main
    device.comms.device.read(endpoint, transfer_buffer, 3000)
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 1021, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 228, in setup_request
    intf, ep = self.get_interface_and_endpoint(device, endpoint_address)
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\usb\core.py", line 244, in get_interface_and_endpoint
    raise ValueError('Invalid endpoint address ' + hex(endpoint_address))
ValueError: Invalid endpoint address 0x81

and the output of 
C:\Users\User>gf info -H
Host tools info:
        host module version: 2021.2.1
        pygreat module version: 2021.2.1
        python version: 3.10.4 (tags/v3.10.4:9d38120, Mar 23 2022, 23:13:41) [MSC v.1929 64 bit (AMD64)]

        module path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet
        command path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet\commands
        gnuradio-companion block path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet\gnuradio

Found a GreatFET One!
  Board ID: 0
  Firmware version: v2021.2.1
  Part ID: a0000a306d4f6a
  Serial number: 000057cc67e632725257

C:\Users\User>gf info -a
Host tools info:
        host module version: 2021.2.1
        pygreat module version: 2021.2.1
        python version: 3.10.4 (tags/v3.10.4:9d38120, Mar 23 2022, 23:13:41) [MSC v.1929 64 bit (AMD64)]

        module path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet
        command path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet\commands
        gnuradio-companion block path: C:\Program Files\Python310\lib\site-packages\greatfet-2021.2.1-py3.10.egg\greatfet\gnuradio

Found a GreatFET One!
  Board ID: 0
  Firmware version: v2021.2.1
  Part ID: a0000a306d4f6a
  Serial number: 000057cc67e632725257
  APIs supported:
    core:
      get_available_classes -- Fetches the available class numbers.
      get_available_verbs -- Fetches the available verb numbers for a given class.
      get_class_docs -- Fetches for documentation the given class.
      get_class_name -- Fetches the string name for the given class.
      get_verb_descriptor -- Fetches the information about the given verb.
      get_verb_documentation -- Fetches the given verb's documentation.
      get_verb_in_param_names -- Fetches the given verb's in-param names.
[... 142 lines cut ...]
      poke -- Writes a raw LPC4330 memory address; for debug.
      read_dmesg -- Fetches the content of the device's debug ring (log).
    dac:
      initialize -- Initializes the DAC driver. Should be called before anything
      set_value -- Sets the DAC value.
      set_voltage -- Sets the DAC value by voltage in millivolts.
    clock_gen:
      output_clock -- Provide the target clock on a given CLKOUT pin,
    adc:
      read_samples -- Initialize the specified ADC for usage, with the given param
      stop_periodic_read -- Stop any active periodic read.
      stream_periodic_read -- Schedule a periodic ADC read, and stream its results to the

C:\Users\User>gf dmesg
[------------] System started after hard reset / power cycle.
[      126499] System clock bringup complete.
[      126530] Configuring board pins...
[      126583] GreatFET initialization complete!
[   182057695] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   201550884] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   210392378] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   256794954] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   373056795] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   380556697] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000
[   388804469] jtag: using half-period delay of 0 microseconds to achieve frequency of 405000

straithe, 2024-08-15:
Since this issue has been opened libusb on Windows has changed quite a bit. @the-lifting-engineer can you please test your setup again and see if you are experiencing the same issue?