greatscottgadgets/greatfet #395: build failing on newer python versions

gozu42, 2021-12-01:
same issue as https://github.com/greatscottgadgets/libgreat/issues/27, see there for details.

```
greatfet/host $ python setup.py build
WARNING: The wheel package is not available.
Traceback (most recent call last):
  File "/home/user/greatfet/host/setup.py", line 34, in <module>
    setup(
  File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib64/python3.10/distutils/core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "/usr/lib/python3.10/site-packages/setuptools/dist.py", line 455, in __init__
    _Distribution.__init__(self, {
  File "/usr/lib64/python3.10/distutils/dist.py", line 292, in __init__
    self.finalize_options()
  File "/usr/lib/python3.10/site-packages/setuptools/dist.py", line 807, in finalize_options
    ep(self)
  File "/usr/lib/python3.10/site-packages/setuptools/dist.py", line 814, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/home/user/greatfet/host/.eggs/even_better_setuptools_git_version-1.0.7-py3.10.egg/even_better_setuptools_git_version.py", line 83, in validate_version_config
    if not isinstance(config, collections.Mapping):
AttributeError: module 'collections' has no attribute 'Mapping'
```

straithe, 2022-03-07:
I'm closing this issue as it is a self-reported duplicate issue.

gozu42, 2022-03-07:
reason for opening it in both "greatfet" and "libgreat" was that aiui it needs to be fixed in both repos.
the fix is likely going to be the same on both sides, so i refed the libgreat one because any talking and figuring out can be done on just one side.

closed 2023-01-11 by antoinevg