greatscottgadgets/greatfet #415: shell variable is NoneType?

unixb0y, 2023-05-22:
I just installed `greatfet` in a new virtual environment.
Launched it via `greatfet shell` and as soon as I hit tab, enter or an opening bracket, it seems to try doing autocompletion, only to subsequently fail due to some object ('shell'?) being Null.

I'm running Python 3.11.2 on macOS.

```
[...]
  File "[redacted]/venv/lib/python3.11/site-packages/IPython/terminal/shortcuts/filters.py", line 170, in readline_like_completions
    return shell.display_completions == "readlinelike"
           ^^^^^^^^^^^^^^^^^^^^^^^^^

Exception 'NoneType' object has no attribute 'display_completions'
Press ENTER to continue...
```

eurocaralex, 2023-05-25:
I'm on the same. Did you find the solution?

pickeditmate, 2023-06-17:
Can you try a lower python version like 3.8 or 3.9 in your venv and see what happens?

unixb0y, 2023-06-19:
> Can you try a lower python version like 3.8 or 3.9 in your venv and see what happens?

I don't use pyenv and don't see why to install a deprecated version of Python just for this.
According to [source](https://en.wikipedia.org/wiki/History_of_Python), even 3.10 full support has ended in May.
Full support for 3.8 and 3.9 ended in 2021 and 2022, respectively.

unixb0y, 2023-06-19:
EDIT!! I was able to get it running on Python 3.11 by installing from source / latest main branch commit instead of pip. It seems like the latest (pip) release is simply missing a lot of new changes.

```
git clone git@github.com:greatscottgadgets/greatfet.git
cd greatfet

virtualenv -p python3.11 venv
source venv/bin/activate

cd host
pip install .

greatfet shell
```

closed 2024-05-17 by antoinevg