greatscottgadgets/greatfet #417: Create new release

unixb0y, 2023-06-19:
As I realized in https://github.com/greatscottgadgets/greatfet/issues/415, I believe a new release is more than overdue.

unixb0y, 2023-07-16:
ping

straithe, 2023-09-28:
Thank you for the ping. We have seen this request and have a couple of PRs and other changes to make before we cut a new release. We really appreciate your patience.

unixb0y, 2024-05-17:
Still no release, it's been 11 months since creation of this issue. And the bug that doesn't even let you use the device (if you install `greatfet` from `pip`) is still there, too, despite the issue having been closed today.
https://github.com/greatscottgadgets/greatfet/issues/415

antoinevg, 2024-05-20:
Just a few things left to do. I'll be pushing a release soon.

antoinevg, 2024-05-22:
Released: https://github.com/greatscottgadgets/greatfet/releases/tag/v2024.0.0

closed 2024-05-22 by antoinevg