greatscottgadgets/greatfet #429: Error when compiling firmware

ikarus23, 2024-05-21:
Hi. I get an error when compiling the firmware:
```
/programs/greatfet/firmware/greatfet_usb/classes/glitchkit/usb.c: In function 'glitchkit_usb_verb_control_in':
/programs/greatfet/firmware/greatfet_usb/classes/glitchkit/usb.c:113:9: error: implicit declaration of function 'delay_us' [-Wimplicit-function-declaration]
  113 |         delay_us(100 * 1000);
      |         ^~~~~~~~
```

I'm using Arch:
```
arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/14.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-14.1.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/arm-none-eabi-gcc/-/issues --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 14.1.0 (Arch Repository) 
```

antoinevg, 2024-05-21:
I don't have an arch box here to test on at the moment but I'd start by checking to see if there are any issues with the versions of `gcc-arm-none-eabi` and `libnewlib-arm-none-eabi` that ship with the distro.

Ref: https://greatfet.readthedocs.io/en/latest/getting_started_firmware_development.html

antoinevg, 2024-05-21:
Ah, it could also be that your particular compiler version has promoted implicit declaration warnings to errors.

I've just pushed https://github.com/greatscottgadgets/greatfet/pull/431/files if you want to try it out.

ikarus23, 2024-05-21:
Perfect! That fixes it. Looking forward to see this in main

closed 2024-05-21 by ikarus23

antoinevg, 2024-05-21:
Fixed in: https://github.com/greatscottgadgets/greatfet/pull/431

Tx @ikarus23 !