greatscottgadgets/greatfet #435: gf shell is not passing python globals into user defined functions

meeas, 2024-08-06:
This appears only to be an issue in `2024.0.0` and `2024.0.1`.

For instance, open `gf shell` and paste this code:

```
import time
def read(x):
    time.sleep(x)
    print("Done")
read(2)
```

This fails with a `NameError: name 'time' is not defined` error.

mossmann, 2024-08-07:
This appears to have been introduced in #414, though I haven't tried to thoroughly test the previous commit because the bug fixed by #414 complicates that testing.

mossmann, 2024-08-07:
I was able to test 4d56571 (prior to #414) successfully by downgrading ipython:
```
pip install "ipython<=8.10"
```
This bug was definitely introduced in #414.

mossmann, 2024-08-07:
The origin of this bug is really here: https://github.com/ipython/ipython/issues/13966
Our current problem is with the workaround that was accepted for this ipython bug.

mossmann, 2024-08-07:
As a workaround, I recommend using `ipython` directly instead of `gf shell`:
```
# ipython
Python 3.11.4 (main, Dec  7 2023, 15:43:41) [GCC 12.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.26.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from greatfet import GreatFET

In [2]: gf=GreatFET()
```

mossmann, 2024-08-07:
Another (horrible) workaround that can be done inside `gf shell`:
```
In [14]: import time

In [15]: def test():
    ...:     print(time.time())
    ...: 

In [16]: test()
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Cell In[16], line 1
----> 1 test()

Cell In[15], line 2, in test()
      1 def test():
----> 2     print(time.time())

NameError: name 'time' is not defined

In [17]: globals().update(locals())

In [18]: test()
1722996533.3737051
```
The truly horrible thing is that you have to call `globals().update(locals())` after every time you update globals.

mossmann, 2024-08-07:
It seems that this is an ancient ipython bug: https://github.com/ipython/ipython/issues/62

We're running into it because https://github.com/ipython/ipython/issues/13966 was not fixed.

mossmann, 2024-08-07:
I think our solution may be to use `start_ipython()`. This test script works:
```
#!/usr/bin/env python3

import IPython
from traitlets.config import Config

c = Config()
c.InteractiveShellApp.exec_lines = [
    'from greatfet import GreatFET',
    'gf=GreatFET()',
]
IPython.start_ipython(config=c)
```

PR #437 (2024-08-07): Fix gf shell not being able to access globals inside user defined functions.
This PR fixes an issue introduced in #414 where executing the following code in the GreatFET shell will fail with a `NameError: name 'time' is not defined error`:

```python
import time
def read(x):
    time.sleep(x)
    print("Done")
read(2)
```

The fix has been tested on IPython versions `8.12.3` and `8.26.0`.

This PR also rolls back the version of IPython specified in the package dependencies to `"ipython~=8.12"` as this was the final version supported by Python 3.8 which is our minimum Python specification for t [... cut]
closed 2024-08-07 by antoinevg