greatscottgadgets/greatfet #445: No documentation or tutroials to set up USB Proxy

DavyPr1tz, 2025-03-11:
It seems to be one of the main features of the GreatFET that it can perform a USB proxy/man-man-on-the-middle but try as I may, I cannot find any tutorial and the Readthedocs page doesn't seem to prove useful. The usb-capture command doesn't seem to work properly, outputting 

"Traceback (most recent call last):
  File "/usr/local/bin/greatfet_usb-capture", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/kali/.local/lib/python3.11/site-packages/greatfet/commands/greatfet_usb_capture.py", line 275, in main
    buffer_size, endpoint = device.apis.usb_analyzer.initialize(args.speed, timeout=10000, comms_timeout=10000)
                            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/.local/lib/python3.11/site-packages/pygreat/comms.py", line 987, in __getattr__
    return self.__dict__[name]
           ~~~~~~~~~~~~~^^^^^^
KeyError: 'usb_analyzer'"

when I run "gf usb-capture -o filename.bin" 

I am really a noob when it comes to the GreatFET One, but I don't think it should be this hard for a beginner to really get a good grasp.
If anyone knows really where to get started, or some projects that could help me understand how to use the device or the Python API, that would be really appreciated.
God bless, and thank you

antoinevg, 2025-03-19:
A good place to start for USB Proxy would be the [Facedancer - "Using USB Proxy"](https://facedancer.readthedocs.io/en/latest/using_usb_proxy.html) documentation. If you haven't worked with Facedancer before I do recommend that you also give the ["Getting started with Facedancer"](https://facedancer.readthedocs.io/en/latest/getting_started.html) section.

The `usb_analyzer` errors you are getting occur because the USB Analyzer functionality requires a GreatFET Rhododendron Neighbor board. Rhodadendron was an experimental design developed by GSG Labs. However, this design was superseded by the [GSG Cynthion](https://greatscottgadgets.com/cynthion/) which is an all-in-one tool for building, testing, monitoring, capturing, and experimenting with USB devices. 

If you would like to continue experimenting with USB and GreatFET the original design for the Rhodadendron Neighbor can be found in the following repository: https://github.com/ktemkin/greatfet-rhododendron"

closed 2025-03-19 by antoinevg

DavyPr1tz, 2025-03-19:
Cool. Thank you for the help!